AllYarnsAreBeautiful/ayab-hardware #4: License missing?

DerAndere1, 2019-04-10:
With the AYAB software being GPL3.0, would'nt it make sense to also add a license to the ayab-hardware repository? It could become licensed under the terms of the GPL 3.0 like Arduino hardware or you could choose something more permissive which works well for hardware (e.g. Creative Commons Attribution 4.0 International License, [CC BY 4.0](https://creativecommons.org/licenses/by/4.0/legalcode) ? Currently you as copyright holders have all rights reserved (https://help.github.com/en/articles/licensing-a-repository ) so the content of this repository cannot be used by anyone without your explicit permission.

chris007de, 2019-04-11:
Back then, we intended to use CC-BY-SA 4.0 for the AYAB Shield, and I thought that we had documented this somewhere (but obviously this is not the case, as I struggle to find the place).
But I hesitate to just add the license for the whole repository now, as it also contains the data for the EMSL AYAB Interface (folder Shieldv2x).
So either we agree on a common license for both pieces of hardware or we have to split into two repositories.
@oskay what are your thoughts on this? Which license did you intend for the EMSL interface?

oskay, 2019-04-11:
The EMSL AYAB interface boards are CC-BY-SA 4.0, which I used since I believed that was the chosen license for the AYAB project. However, it is my copyright to assign, so I'm happy to license it under other terms (in addition or instead of those) if you prefer. You have my full permission to set the license type, as you see fit, for that set of hardware designs.

Also, GPL is a kind-of iffy license structure for hardware. The TAPR OHL or CERN OHL may be better models to start with.

chris007de, 2019-04-14:
We'll stick to the CC-BY-SA 4.0 license for now.

closed 2019-04-14 by chris007de