AllYarnsAreBeautiful/ayab-hardware #5: Shield V2 Footprints U2 and U3 in Gerber files do not match SSOP20

atrophy, 2019-08-08:
The footprints on the current version of the Gerber files don't match the SSOP20 part listed in the BOM. Should we be using the SOIC version of the PCF8574?

[image]

oskay, 2019-08-08:
Yes, looks like that's an error on the BOM.

oskay, 2019-08-08:
Are you fabricating boards to sell?

atrophy, 2019-08-08:
Not to sell no, I work at Fab Lab Wgtn, part of Massey University and we're converting a couple of the university's knitting machines over to use in the lab.

oskay, 2019-08-09:
(That's fine -- we're still hoping to find someone happy to make and sell a batch.)

chris007de, 2019-08-11:
Thanks a lot for fixing this issue!

closed 2019-08-11 by chris007de