AllYarnsAreBeautiful/ayab-hardware #8: Feature suggestions for AYAB interface

t0mpr1c3, 2019-11-23:
Eventually I would like to re-design the AYAB interface. Some improvements that I could suggest are straightforward (e.g. longer USB cable). Others incremental (e.g. migrating from Adafruit Mini Metro to something cheaper, 1x MCP28017 instead of 2x PCF8574, common bus resistor networks). In my view the most important area for new development is revamping the enclosure. But I will need to play around a good deal before coming to any conclusions on that.

Does anybody have any thoughts or feelings that they would like to chip in?

chris007de, 2019-11-23:
@oskay @andz82

oskay, 2019-11-24:
We have made some prototype enclosures, that worked pretty well, but weren't able to go to the next stage with that.  @t0mpr1c3 we can chat about that offline.

There's not much issue with changing to a lower-cost USB/micro interface, but take care that it does not accidentally increase the cost instead.

I would be *very* careful about going to a different I/O expander. The AYAB software has to support every version of the AYAB hardware, and adding *yet another* chipset to support should only be done if there is a **clear and compelling reason**.

t0mpr1c3, 2019-11-24:
Fair point about the I/O expander. Happy to chat offline about enclosures.

An attractive suggestion mentioned to me today was to reserve a pin on the microcontroller for a row-counting Hall effect sensor.

t0mpr1c3, 2021-08-03:
- Comparators are going to be faster than ADC for evaluating the Hall sensor output.
- Eventually, it will become desirable to support USB-C. I would not be surprised if Apple are planning to phase out USB-A on their laptops. USB-C is already standard on iPads and phones, which we may want to support some day.
- A chip with more IO pins would not require I2C expanders.

chris007de, 2021-08-07:
Hi Tom, 

> * Comparators are going to be faster than ADC for evaluating the Hall sensor output.

We have to keep in mind that there obviously are different hall values for different machine types, therefore these comparators would have to be compatible to either levels (https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/master/settings.h#L49)

> * Eventually, it will become desirable to support USB-C. I would not be surprised if Apple are planning to phase out USB-A on their laptops. USB-C is already standard on iPads and phones, which we may want to support some day.

Good point...it may be worth to check. I like the USB-C port as well for it's rigidness.
Still, if a host device currently has USB-C, there is still the possibility to use a USB-C to Micro-USB cable for now. USB-C <-> USB-C cables are still pretty rare in the wild (at least in my bubble of colleagues, family and friends).

> * A chip with more IO pins would not require I2C expanders.

Guess that's probably a decision of costs...bigger chip might cost more than I2C expanders.

oskay, 2021-08-07:
USB C support is trivial; that's just a cable type, and it's straightforward enough to switch to a USB C cable. The real concern is that future devices may, at some point, not support USB 2.0. Thus far, newer versions of USB (including properly built USB 3 cables) do generally support USB 2.0 for backwards compatibility.

t0mpr1c3, 2021-08-07:
> We have to keep in mind that there obviously are different hall values for different machine types, therefore these comparators would have to be compatible to either levels (https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/master/settings.h#L49)

Yes, a disadvantage is that you would need one comparator (and hence one pin) per level. The data acquisition would be basically instantaneous, whereas reading the ADC takes about 0.1ms. Although I suspect UART communication between the Atmega and its USB bridge is the major bottleneck in terms of response time.

t0mpr1c3, 2021-08-07:
> USB C support is trivial; that's just a cable type, and it's straightforward enough to switch to a USB C cable. The real concern is that future devices may, at some point, not support USB 2.0. Thus [...]

This is good to hear. I hate micro USB, the cables are flimsy and always breaking. A real downgrade from mini USB.

oskay, 2021-08-07:
The micro connectors are designed to last for twice as many cycles as mini:
https://en.wikipedia.org/wiki/USB_hardware#Durability

t0mpr1c3, 2021-08-09:
It's not the connection that fails, it's the cable itself.

Rixstaa, 2023-02-20:
I would like to have a option to use a different motor to make all carriages electronic with some switch on each end for it to go left to right with out doing it manualy. Would also like to know if theres a different hall position sensors we can use as originals are becoming hard to find, maybe schematic to make some etc or already a arduino one that will work? I was even thinking adding a tft screen would be nice for some check functions.

t0mpr1c3, 2023-02-20:
There isn't anything complicated about the Hall sensors as far as I can
remember. Remanufacturing them would be mostly just trial and error to get
the sensitivity right.

closed 2024-02-28 by dl1com