AllYarnsAreBeautiful/ayab-hardware #10: beeper circuit

t0mpr1c3, 2021-01-13:
The specified beeper has a coil resistance of 47 Ohms and draws 40 mA (max) through the collector of a MMBT3904 transistor and is connected to the 5 V rail through a 62 Ohm resistor.

Ignoring for the moment the Vce of the transistor and the coil inductance, that means the voltage drop across the resistor should be about 3 V and the voltage drop across the beeper is about 2 V. That's lower than its supported range of input voltages (3-7). For what it's worth, I have tried both the specified part and some cheap alternatives and they have all worked.

I haven't played around with any alternative circuits but I believe the typical application would include a flyback diode to protect the transistor. Again, I haven't noticed any problems so far.

Any thoughts?

oskay, 2021-01-13:
The most important thing is to match the beeping frequency to the device resonant frequency, to get enough loudness. Different people have different opinions about how loud a beeper should be.

t0mpr1c3, 2021-01-13:
I think I will experiment a little bit. I tend to think louder is better, as there will be a silent option in the software going forward.

oskay, 2021-01-13:
Possible option to consider for the future: Migrate to making noises in the software, not the hardware. Computers have headphone ports and things; can be made really loud or quiet, easily.

t0mpr1c3, 2021-01-13:
Yes, that will be implemented in the software in future and will make the beeper redundant. I really don't think there is any advantage to having both. Can you think of any reason to retain the beeper?

Apart from anything else, some users may have strong preferences on the subject of electronic noises and I tend to think these are more easily addressed in software.

oskay, 2021-01-13:
This is probably a question that should be brought to the userbase as a whole.

t0mpr1c3, 2021-01-14:
Yes. There will be user input to the software update and that should clarify things.

VIPQualityPost, 2024-02-27:
https://github.com/AllYarnsAreBeautiful/ayab-firmware/pull/143
beeper configuration implemented, seems issue is closed

closed 2024-02-27 by VIPQualityPost