AllYarnsAreBeautiful/ayab-hardware #17: Machine selection

t0mpr1c3, 2021-10-04:
For v1.0, the plan is to select the machine from a drop-down menu instead of it being hard-coded in the firmware.

Another solution would be to have machine type hard coded on the shield. The obvious way to do this would be to connect a resistor divider to an analog pin. If the shield is intended to be used with more than one type of machine, this could be done using a DIP switch and resistor ladder.

When the desktop software starts up, it can check the voltage on the pin to determine the machine type. If no resistor is connected, software selection can proceed.

chris007de, 2021-10-05:
Do I understand correctly that you propose this to be an additional (/optional) way for automatic machine type identification?
So if the software does not recognize a machine type automatically, it falls back to manual selection in software?

From my experiences, having as few possible points of failure/misconfiguration is desirable. By having software selection only, it is pretty obvious where to set things up. When having a hardware switch (reconfigurable DIP switch without need to solder would be a must-have in my opinion), things already get a bit more complicated, as somebody has to check things on the AYAB interface as well if somethings goes wrong.

t0mpr1c3, 2021-10-05:
I understand your concern about making things more complex than they need to be. I just sort of improvised the DIP switch idea, thinking that people might want boards that can be used with multiple different machines. Maybe that is a bad idea, and just staying with software selection makes more sense in this case.

For people with existing boards, nothing would change: selection would be done in software.

For new boards, machine type would be automatically detected, and the machine type options in the preferences dialog would be greyed out. It actually makes things simpler for users, who would no longer be able to select the wrong machine in the preferences.

The hardware changes are minimal. v2 shields are already machine-specific: they have different covers and connectors depending on machine type. This change would just involve a couple of extra resistors attached to one pin. If software selection is preferred (perhaps because the user possesses adapter cables that can connect the shield to another type of machine), then these resistors can simply be omitted from the circuit board, enabling software selection of the machine type.

VIPQualityPost, 2024-02-27:
It seems like we didn't address this in the new design, but also having it assembled by third party makes it annoying to tinker with the BOM for each potential machine/customer. 

This does remind me we should revisit the OTP/ hardware revision programming soon though, certainly before we push the ESP32 design out of prototype.

t0mpr1c3, 2024-02-27:
Setting the machine in software seems good enough for now, I agree with Chris that a jumper is probably all we need to set it in hardware if we want to go that route.

VIPQualityPost, 2024-02-28:
OK, on the current rev of ayab-esp32, we only have 3 unused pins, these could be used for hardware ID (8 possible machines) but imo this is not super necessary and makes it difficult/impossible to use those pins later if we need to change things down the line. I would suggest that we just stick with sw machine identification for now.

t0mpr1c3, 2024-02-29:
Currently we have 3 hardware types and we can define that using one pin and one jumper.
0 = no jumper
1 = jumper between pin and ground
2 = jumper between pin and 3V

So we could break out one spare pin plus ground and 3V to a 3 pin header and have 2 spare pins for other stuff.

VIPQualityPost, 2024-02-29:
Oh, I totally misunderstood this as Brother machine identification, not AYAB hardware... There are already firmware defines to the new platform, we can just add something there and also have it return something in the identification step of the communication with some hex ID.

VIPQualityPost, 2024-02-29:
Unless you thnk the next generation AYAB hardware will also be build on ESP32-S3 this seems like the best way to handle it I think.

t0mpr1c3, 2024-02-29:
Yes. So do we agree that we'll do the machine selection in software, and the board will have some kind of identity defined in firmware?

VIPQualityPost, 2024-02-29:
Yes, I agree.

t0mpr1c3, 2024-03-01:
Closing as nothing is required for hardware.

closed 2024-03-01 by t0mpr1c3