AllYarnsAreBeautiful/ayab-hardware #18: What software is used for schematic ?

OddmarDam, 2021-10-11:
Hi.

What software is used for this schematic ? Is it Eagle or Kicad or something else ?

chris007de, 2021-10-12:
> http://www.geda-project.org/

That's for Shieldv2x ("AYAB Interface"). For Shieldv1x ("AYAB Shield") it's eagle.

> Unfortunately this software is no longer actively maintained. I use the Ubuntu 16.04 version.

For this reason, I'd propose to move to KiCAD sooner or later, as it's actively developed FOSS.

t0mpr1c3, 2021-10-12:
> That's for Shieldv2x ("AYAB Interface"). For Shieldv1x ("AYAB Shield") it's eagle.
>
>
> For this reason, I'd propose to move to KiCAD sooner or later, as it's actively developed FOSS.

I agree. I hear good things, I just haven't got around to learning how to use it.