AllYarnsAreBeautiful/ayab-hardware #19: Read all relevant signals for G carriage detection

dl1com, 2023-03-18:
see https://github.com/AllYarnsAreBeautiful/ayab-firmware/issues/51#issuecomment-1467086056 and following

VIPQualityPost, 2023-07-17:
So what is the resolution to this?
Do we want to have two input pins for 910 EOL, one for K and one for L? 
Or should we just OR the two signals, because we can sense if the carriage has moved >3 needles but shows a pulse longer than single magnet?

maybe we can get some comments from @jpcornil-git

jpcornil-git, 2023-07-17:
I would use both (digital) inputs to avoid another (less robust ?) detection mechanism for LKG carriages.

One input is indeed equivalent to "hallValue < FILTER_R_MIN" and the other to "hallValue > FILTER_R_MAX" and existing code can therefore easily be modified to cope with this.

jpcornil-git, 2023-07-17:
... and you can't differentiate K and L anymore with an OR approach (one short pulse for both) -> not an option anyway

VIPQualityPost, 2023-07-17:
Right, I was thinking because they are a pulse that you can do it, but it doesn't work for other machines anyway. It will have it's own pin.

VIPQualityPost, 2024-02-27:
We did it!

closed 2024-02-27 by VIPQualityPost