AllYarnsAreBeautiful/ayab-hardware #20: [WIP] Requirements for Next-Gen Hardware

dl1com, 2023-06-19:
# Introduction

As discussed in the developer meetings and on Discord, we want to go for a new AYAB hardware more sooner than later.
Here is a collection of requirements. This serves as a common ground for discussion in Discord and developer meetings.
Once all (or at least the most significant) topics are consolidated, this issues serves as meta issue and the requirements are split into separate issues/tasks.

Many of this is work in progress and to be discussed (marked with a ❓). Once it seems that there is a common opinion for a topic, the ❓ will be removed.
Design decisions to be made are marked with a ⚖️.
This first post will get updated continuously to reflect the current state of discussion.

# Requirements and Design Decisions

## Form factor / Dimensions

- ❓The board shall have the same dimensions as the EMSL AYAB Interface to be able to be fitted into the machine
- ❓The components and connectors shall be placed as similar as possible as on the EMSL AYAB Interface to ensure its compatibility to the cable lengths and the space in the machine.

## High level

- The board shall enable the usage of the current AYAB APIv6 over a USB Serial connection
- The board shall provide Wifi connectivity to enable a future shift to a wireless connection from the AYAB Desktop software
- The board shall provide enough ressources (i.e. flash memory, RAM) to be able to serve a web interface (i.e. written in a JS framework like vue.js) which reflects the functionality of AYAB Desktop v1.0

## Power Supply

- ❓The board shall use an external power supply to enable usage in machines where the original power supply is broken
- ❓⚖️The board shall use a standard interface for power supply, such as a barrel connector OR USB-C.
- ❓⚖️ The board shall/shall not provide an On/Off switch

## Controller

- ❓The board shall use a microcontroller with integrated Wifi connectivity
- ❓⚖️ The board shall use an ESP32 microcontroller as it is very well supported and available. Numerous examples show that it is capable of serving web applications hosted on the controller via Wifi. (other suggestions? Pi Pico /w Wifi?)
- ❓⚖️ The controller shall be fitted as a daughter board (i.e. ESP32 DevKitC; like the MiniMetro on the EMSL AYAB Interface) OR the controller shall be soldered directly onto the board

## Components

- ❓The board shall use "common available" components, i.e. no "exotic" parts, so the probability of missing parts/obsolenscence is reduced.
- ❓⚖️The board shall use only/mostly parts available in the **<insert PCBA manufacturer here>** PCBA library, so it is possible to order fully assembled and tested board at this manufacturer.
- ❓⚖️ The board shall provide the same LED indicators as the EMSL AYAB Interface
- ❓⚖️ The board shall/shall not provide a hardware buzzer/speaker

## Expandability

- ❓ The board shall a generic interface for future extensions (i.e. display), i.e. expose the I2C pins to a connector.

## Bugfixes / Lessions learned from earlier versions

- The missing sensor signal for proper carriage detection has to be connected (see https://github.com/AllYarnsAreBeautiful/ayab-hardware/issues/19)

## Software Interface

- The board has to be supported by the AYAB firmware
- The board has to be supported by the AYAB Desktop software

## Organizational decisions & Tooling

- The board shall become OSHWA certified
- The board shall be designed using KiCAD 7
- ❓⚖️ The board data shall be hosted in ayab-hardware (new folder) OR in a new, separate repository?
- The repository shall have a GitHub action for automatic generation of HTMLBom
- The repository shall have a GitHub action for automatic generation of KiDiff visual diff

# Design draft

Made by @jpcornil-git 
[image]

t0mpr1c3, 2023-06-30:
Possible serial EEPROM e.g. 24FC type to store hardware information. (https://github.com/AllYarnsAreBeautiful/ayab-firmware/issues/106)

VIPQualityPost, 2024-02-27:
I think we hit all of these things, with the exception of generating production files (gerber, BOM, CPL).
However, with the new release of Kicad 8, there is improved kicad-cli support for generating these files, so we can probably set up CI for this.

closed 2024-02-28 by dl1com