AllYarnsAreBeautiful/ayab-hardware #28: Prototype board revisions

VIPQualityPost, 2024-02-26:
- fix BOM from XL1509-5.0 to XL1509-ADJ
- determine why MCP I2C pullups are missing
- increase trace width out of 5V/3V3 regulators
- User I2C port is vertical, find horizontal part and fix footprint
- 12V rail showing 4V when only plugged in via USB, determine if we need to fix this (jp identified as protection diode in XL1509)
- Adjust LED resistor values to match brightness across colors

t0mpr1c3, 2024-02-27:
Apart from the trace width from the regulators, did you notice any thermal issues? I generally just throw in lots of vias and hope for the best.

VIPQualityPost, 2025-02-10:
I think we can mark this closed since bringup is done, and instead handle the new requested changes in #38 .

closed 2025-02-10 by VIPQualityPost