AllYarnsAreBeautiful/ayab-hardware #36: Current state of ayab-esp32

thoka, 2024-03-20:
We just inherited a Brother KH910 and would order some PCBs and try to build an interface for it.
 
Is the new interface board suitable to get it running already or would you recommend to wait for a new revision?

VIPQualityPost, 2024-06-11:
Hey! Sorry for missing your message....
I am still working on validating the new ESP32 boards, they are not quite ready for production yet. If you want to start using your machine _today_ then I would recommend that you spring for the arduino shield or the interface (long black board). However, if you are patient, I'll be done with bringup and testing in the next week or two, and after AYAB 1.0 is out, we will work on merging in support for the new hardware.

bfdkaz, 2024-12-28:
hi any progress on this. looking at buying a machine but want an interface rather than shield. thanks

VIPQualityPost, 2025-02-10:
We still didn't push out v1.0 yet, this is the reason that nobody started on firmware for this board yet, so nothing is ready still. We now launched beta5, there shouldn't be much time now until the formal release and then some work will get started again on the esp32 boards.

closed 2025-02-10 by VIPQualityPost

Maitresinh, 2025-02-19:
I've just send it to JLBpcb.... :/
It's way i could find the ESP32 script....