AllYarnsAreBeautiful/ayab-hardware #42: Wrong part number for I2C port expander in AYAB interface BOM

dl1com, 2024-11-06:
In https://github.com/AllYarnsAreBeautiful/ayab-hardware/blob/main/ayab-interface/assembly/ayab_rs_oshw.csv#L17 the Manufacturer Part Number and order numbers are stating a wrong version of the chip (PCF8574A).

The correct one is PCF8574 (without trailing A).
The parts have different I2C adresses and are therefore not interchangeable, so it is crucial to fix this in order to avoid DoA interfaces.

VIPQualityPost, 2025-02-10:
Does this require any changes on the actual design in geda or can we just update the BOM file? 
Is it worth maintaining interface design files? It doesn't appear you can easily import geda project to kicad, and geda seems generally poorly maintained, the last release is many years ago at this point.

VIPQualityPost, 2025-04-01:
@t0mpr1c3 Can you address above? If it's only a BOM change this is easy, otherwise I don't really care to try and modify the gEDA design...

t0mpr1c3, 2025-04-14:
>  @t0mpr1c3 Can you address above? If it's only a BOM change this is easy, otherwise I don't really care to try and modify the gEDA design...

Yes, it's just a typo in the BOM.

closed 2025-04-18 by VIPQualityPost