AllYarnsAreBeautiful/ayab-hardware #43: ayab-esp32: KH910 digital right Hall sensor signals do not match what the board comparators generate

jonathanperret, 2025-01-02:
## Summary

On the `ayab-esp32` board, the KH910 digital outputs for the right Hall sensor are mixed in with outputs from the board's comparators (which are not meant to be active on a KH910). But the signals coming from the machine's built-in comparators are not compatible with those from the board's, meaning that firmware would have to interpret them differently based on the machine type.

Also, it appears that the board's comparator outputs will both sink current when the analog sensor output is left open (as would be the case on a KH910), making it impossible for the machine's digital outputs to override the board's comparator outputs.

Note that while I have measured the machine's signals on an actual KH910, the `ayab-esp32` board's behavior I posit below is only based on my understanding of the schematic and datasheets, so I apologize in advance if I made a mistake.

## Context

Here's what the comparator setup for the right Hall sensor looks like on the `ayab-esp32` board as of commit 6ace2960d:
[image]

If I am reading the [LM393 datasheet](https://www.ti.com/lit/ds/symlink/lm393.pdf) correctly, this should result in the following behavior for `EOL_R_K` and `EOL_R_L`:

| EOL_R | EOL_R_K | EOL_R_L |
| --- | --- | --- |
| < 0.88V  | Low | Hi-Z |
| 1.6V | Low | Low |
| > 2.94V  | Hi-Z | Low |

So far, so good when used on a machine that has an analog right Hall sensor output such as the KH930.

## KH910 situation

Because the KH910's connector for the right sensor exposes digital outputs from its built-in comparator (see https://github.com/AllYarnsAreBeautiful/ayab-hardware/issues/19), the board's comparators cannot be used.

Here is the behavior of the machine's built-in comparator outputs as measured on a KH910 (in **bold**, values that differ from the board's behavior):

| EOL_R | EOL_R_K | EOL_R_L |
| --- | --- | --- |
| < 0.88V  | **Hi-Z** | Hi-Z |
| 1.6V | **Hi-Z** | Low |
| > 2.94V  | **Low** | Low |

It appears the machine's comparator for the `K` magnet is wired in reverse of the board's, causing it to sink current only when a `K` magnet is present. As a result, firmware reading a **Low** value on `EOL_R_K` cannot determine whether a `K` magnet is present or not without taking into account the machine type.

Additionnally,  because the `EOL_R` input is left floating when connected to a KH910, it is driven to 1.6V by the biasing resistors, causing both of the board's comparators to go **Low** and sink current. This means that the machine's comparators can never cause the firmware to see anything other than the absence of a magnet.

jonathanperret, 2025-01-03:
I wanted to be sure about the 910 measurements that I did a while ago, so I redid them, with pictures so that I can be sure I didn't just write things down incorrectly.

Here's the K carriage passing in front of the right sensor, while the multimeter is monitoring pin 8 of JP5 (no pull-up resistor so the floating voltage isn't relevant, just that it's zero or not):

[image]

As noted above, the pin goes low only when the K carriage is detected.

Here's the L carriage, monitoring pin 7 of JP5:

[image]

Again, matches the table above: pin is low **unless** the L carriage is detected. This is a bit unexpected (which is why I wanted to make doubly sure my readings were correct), it feels like it would have been more logical to have each pin go low when it detects a carriage, but maybe it simplifies the implementation downstream.

jonathanperret, 2025-01-03:
@jpcornil-git confirmed on Discord that it was intended for the comparators to sink current only when the thresholds are passed, i.e. when a magnet is detected, but a misreading of the datasheet caused their inputs to be flipped.

I guess all the comparators inputs should be flipped, so that they output a low signal when they detect a magnet, and float by default, as was intended initially.

This would make the 910's pin 8 (K detector) OK to connect to the comparator output as it is already.

But the 910's pin 7 (L detector) still would need some kind of inverter to be compatible with the comparator output.

Note that you can't just flip the inputs of the L comparator (i.e. have it float when an L magnet is detected, like the 910's pin does), because this would make the comparator force the output to low when its input is floating, preventing the 910's pin 7 from overriding it.

jpcornil-git, 2025-01-04:
I would keep the comparator output high-Z when there is no detection; this makes KH910 and KH930 different for FW (minor adaptation) but allows to share the same IOs for the EOL_R signals.

I prefer to have KH910's EOL_R digital signals routed to digital inputs (with pullup enabled) and analog signals routed directly to analog pins (as of today). Note that this is what you get with the fix for JP5.7 as well (and WiFi with UnoR4).

If deemed required, you could also implement an auto-detection mechanism for KH910/930 (internal pullup on JP5.7);
- high for more than 3 needles = KH910 not connected -> switch to KH930
- low for more than 3 needes = KH910 connected -> switch KH910

For test purpose/confirm behavior, you could also connect JP7 with flying wires to arduino and connect JP5.7 to a spare GPIO (enable pullup on both JP7.7 & 8)

jonathanperret, 2025-01-04:
> Note that I prefer to have KH910's EOL_R digital signals routed to digital inputs (with pullup enabled) and analog signals routed directly to analog pins (as of today). This is what you will get with the fix of the current shield for JP5.7 as well (and WiFi with UnoR4).

I’m confused. If you want to route analog signals directly to analog pins then there’s no need for the whole comparator setup on this board (we’re talking about AYAB-esp32 here). Personally I’ve been wondering why the comparators were there, as it seemed simpler to route the analog signals to ESP32 pins (via a voltage divider to keep them under 3.3v), but I assumed they were there for a reason.

If they don’t need to exist, then I agree it would be best to keep the wiring similar to what we should have on the upcoming fixed Arduino shield: left (and right on non-910) analog signals going to analog pins, and 910 digital signals going to digital pins.

jpcornil-git, 2025-01-04:
Agree, this is what I would do as well, i.e. no comparators but direct connections (GPIO configured as analog input or digital inputs with pullup enabled for KH930/KH910 EOL_R respectively) for existing shield and esp32 (passive level shifters for the latter to scale analog signal from 5V0 down to 3V3).

I believe comparators are there to offer:
- a full digital interface and ease interfacing with upcoming 3V3 MCU (easy to address)
- no differences between KH910 and KH930 from a FW perspective (not true anymore)

Given that they aren't must-have and they break current EOL implementation, I would revert to the shield approach.