AllYarnsAreBeautiful/ayab-hardware #47: Add assembly components to shield BOM

vthunder, 2025-01-06:
The KiCad project for the shield only contains symbols actually placed on the board, but there are a few more required for assembly (eg extension cable, buzzer).

These should be added so they appear on the KiCad BOM.

Maitresinh, 2025-02-19:
Also : i've found the values for the BOM here

https://www.ayab-knitting.com/ayab-shield/

But the link to make the brother connectors (KH1910 or 920) is dead.
Does the components listed in the BOM for the connectors are still the good ones ?

jonathanperret, 2025-02-19:
> Also : i've found the values for the BOM here
> 
> https://www.ayab-knitting.com/ayab-shield/
> 
> But the link to make the brother connectors (KH1910 or 920) is dead. Does the components listed in the BOM for the connectors are still the good ones ?

The precise manufacturer and shop references may be obsolete (or not, I don’t actually know) but I can tell you the part requirements for the shield have not changed since that page was written.

> 
> Edit : also... ULN2803APG,CN is not avaiable anymore. Digikey propose a substitute (TBD62083APG). Is that ok ?

That part looks superficially similar to the ULN2803 but I wouldn’t try substituting it. ULN2803 is a very common part with multiple manufacturers, that should be easy to find in DIP18 format, just search for that. A quick search on Digikey returns an ST part that is in stock: https://www.digikey.fr/fr/products/detail/stmicroelectronics/ULN2803A/599591 — a bit pricey but you only need 2 of those.

Maitresinh, 2025-02-19:
Thanks. 

Regarding how to make the connectors, there is no copy of the dead link on archive.org. If anyone could post it again....

jonathanperret, 2025-02-20:
> Regarding how to make the connectors, there is no copy of the dead link on archive.org. If anyone could post it again....

Sure. The link was broken by #39 which moved the PDF to here: https://github.com/AllYarnsAreBeautiful/ayab-hardware/blob/main/ayab-shield/pdf/kh9xx_power.pdf

You may be disappointed though, this PDF only contains a drawing of the outline of the part you need to make (3D printed or otherwise) to complete the connector.

VIPQualityPost, 2025-04-01:
> You may be disappointed though, this PDF only contains a drawing of the outline of the part you need to make (3D printed or otherwise) to complete the connector.

Should we upload a .step file so that users don't have to draw this (primitive) shape from scratch? I have the part already from when I made a full model of the Brother power connector, I think...

Maitresinh, 2025-04-01:
I've made one.
Here it is : https://cloud.500nuancesdegeek.fr/index.php/s/4n2qd43M9nNtSRH

commit 4411f7a3 (2025-04-07): Fix #47, #46, #45, #44
closed 2025-04-07 by jonathanperret