AllYarnsAreBeautiful/ayab-hardware #49: A bit Confused / A few questions

Maitresinh, 2025-02-16:
A few questions before diving into this wonderfull hack (many thanks by the way) :

**1- Which one to choose** 
"Ayab interface" := the  take from mad scientifst (using adafruit) based on the first Ayab interface using arduino uno, but with a new form factor
     "Ayab Esp32" = the new design from AYAB, adaptating the mad scientist take with an ESP32 instead of Adafruit 

So, Ayab Esp32 should be considered as the latest ayab, using Esp32 for opensource and cost shake.

Am i right ?

**2. BOM**

I've found that
https://github.com/AllYarnsAreBeautiful/ayab-hardware/blob/main/ayab-interface/assembly/ayab_rs_oshw.csv

...but for ayab interface. I suppose the BOM file is into kicad files.

**3.** Installation and soldering

Doc is outdated. I suppose the installation (cables) is the same as for Ayab 1.0

**4. Flashing**

Where is the file/package ? I counted find a "script" folder into the Esp32 ayab folder

Maitresinh, 2025-02-18:
1/ Solved. I used the ESP32 PCB, assuming it's the last one.
2/ Solved. Found the BOM in Kicad

But still wondering with scripts for flashing

jonathanperret, 2025-02-19:
Hi! Sorry for the delay in responding to your questions.

AYAB Interface: you have the nomenclature mostly correct, except that the original (and still updated/manufactured) AYAB hardware is usually called the "AYAB Shield" whereas "AYAB Interface" is reserved for the EMSL remake that includes a microcontroller board (https://wiki.evilmadscientist.com/AYAB_Interface) and has a larger form factor. Those two boards have essentially the same schematic and share the same firmware. As of this writing, the EMSL board is no longer being manufactured by EMSL, and while there are a couple of vendors online that can build and sell them on demand, I believe most new AYAB users currently go for the AYAB shield and an Arduino Uno, probably because it is much easier to DIY.

If you're only interested in building working AYAB hardware, your simplest option would be to get an AYAB shield manufactured from the KiCad project in https://github.com/AllYarnsAreBeautiful/ayab-hardware/tree/main/ayab-shield. It doesn't look like we have an exported BOM file for the shield but the part list is pretty straightforward. Of course you could also just order a kit from https://ko-fi.com/s/716cda7fa9.

If you're interested in building an AYAB Interface specifically, it will be a bit more involved — the fabrication files are in this repository (https://github.com/AllYarnsAreBeautiful/ayab-hardware/tree/main/ayab-interface) but I can't speak to the full validity of the BOM — there's at least one open issue on that: https://github.com/AllYarnsAreBeautiful/ayab-hardware/issues/42.

Regarding AYAB-ESP32 now: this is an entirely new board design that @VIPQualityPost has been working on, based as the name implies on an ESP32 MCU instead of an Atmega328p. You shoud be aware that that design is still in development and has not to the best of my knowledge been used in a knitting machine yet. There's at least [one identified issue](https://github.com/AllYarnsAreBeautiful/ayab-hardware/issues/43) that implies upcoming changes to the design. And no firmware has been written for that board yet.

I've just seen that you've found an older issue about the state of the project (https://github.com/AllYarnsAreBeautiful/ayab-hardware/issues/36#issuecomment-2647007830) so this is probably redundant information at this point.

closed 2025-02-19 by jonathanperret

Maitresinh, 2025-02-19:
thanks for your reply. 

In order to sumarize and check if i understand everything:

- **Ayab shield :** the first one with an arduino. Simple and working
- **Ayab interface** : EMSL version. Kicad files are on the Github under /Ayab interface. Working using the same arduino script than Ayab Shiled
- **Ayab ESP32** : the new version in process. PCB are on the Github, but the scripts are not. Still in progress.

Now the questions about the ESP3 version :
- Is the PCB working ? (i understand it has not been tested yet, but i mean, do we have to expect changes ?)
- Is the an even (very) approximative date for the release of the (first) script ?

-

VIPQualityPost, 2025-04-01:
> Now the questions about the ESP3 version :
> 
> * Is the PCB working ? (i understand it has not been tested yet, but i mean, do we have to expect changes ?)
> * Is the an even (very) approximative date for the release of the (first) script ?

I think there are some plans to start working on firmware for this board very soon! There will be some large changes in the firmware to move to a more flexible architecture (ayab-async) so this will require a good bit of work.

As far as the board is working: yes, it works but there are some changes to be made due to later discoveries (#43 ), I will fix them this week and open a PR.

Maitresinh, 2025-04-01:
ok. thanks for your answer. I will wait for the firmware.
Meanwhile, I've made 10 units of this PCB. If anyone wants one, i can send it by the post (i only charge 8 euros + shipping to cover the cost)