AllYarnsAreBeautiful/ayab-hardware #56: Increased clearance around holes in shield v1.5

t0mpr1c3, 2025-04-15:
I think the holes in the shield are potentially useful to secure the shield on top of the arduino, but only if they have sufficient clearance around them for a washer or pan screw head. The limits of this clearance should be allocated using the courtyard feature.

VIPQualityPost, 2025-04-17:
I don't particularly think they are useful and don't see a way to increase clearance - the board is already too packed with the through-hole parts used. I would rather see them go away but I think opinion on this may be mixed.

t0mpr1c3, 2025-04-20:
I like them but only if there is sufficient clearance. I'm not sure what you can do with a 3mm hole that has 3.2mm of clearance, so getting rid of the holes could be the consensus option.

t0mpr1c3, 2025-05-05:
Close as won't fix.

closed 2025-05-05 by t0mpr1c3