AllYarnsAreBeautiful/ayab-hardware #57: LED resistor values on shield v1.5

t0mpr1c3, 2025-05-03:
Currently the values of the resistors in series with the LEDs on v1.5 of the shield are 150 ohms. This seems right for an LED with a 2V drop on a 5V line since (5-2)/150 = 0.02 or 20 mA. Red and yellow LEDs typically do have a 2V drop, but green LEDs typically have a 3V drop so 100 ohms might be more suitable for that one.

t0mpr1c3, 2025-05-05:
Another issue here is that the pins on the Uno R4 can only source 8mA of current, whereas the R3 could provide 20 mA. So the values of 100 & 150 ohms should be increased to 270 & 390 ohms in that instance.

jonathanperret, 2025-05-05:
The current limitation on R4 pins is a very good point. Sticking to a single target current however isn't the whole story, as different LEDs have a different brightness/current relationship. You can find LEDs that will shine brighter at 5mA than others at 20mA, even within the same color. See https://news.sparkfun.com/5461 for example for one way a device manufacturer approached it.

I think that leaves us with a few options:
1. pick a single target current (e.g. 8mA, probably lower for a safety margin on R4) and pick resistors based on the actual forward voltage of the selected LEDs;
2. pick a single resistor value — like the current design, but higher to make sure the R4's pin current limit isn't exceeded — and try to select LEDs that have a relatively similar brightness at that current;
3. do a bunch of testing of various LED models to select the best set of resistor values for (approximately, as that will depend on the human observer in any case) consistent brightness and write down the selected resistor values and LEDs to the BOM.

I would favor option 2 for the through-hole shield, as ease of assembly is an important factor in that case and minimizing the BOM length helps with that. So, bumping the resistors from 150 Ω to 560 Ω would be enough I guess?

For SMT, professionally assembled boards, the equation is different and I suppose one could do the research that goes with option 3. Not sure that's worth the effort though, maybe option 1 would be enough.

t0mpr1c3, 2025-05-05:
You are right that the brightness, and hence the resistor values, depends on the particular LEDs that are specified.

Typically LED brightness (in mcd) is specified at 20mA but if you dig into the datasheets you can usually find the brightness/current curve. It's generally more or less linear up to 20 mA.

The voltage drop also varies with the current applied. These curves are also in the datasheet.

So if you go by the ratings in mcd you can specify the brightness that you want and calculate the current and forward voltage and get the value of resistor that is required.

It doesn't require all that much experimentation, it's more about choosing suitable LEDs and working backwards from there.

So probably we need to consider whether the LEDs specified in the BOM are suitable (i.e. brightness enough at lower current).

t0mpr1c3, 2025-05-05:
@VIPQualityPost The LEDs that are specified currently, according to the schematic anyway, are HLMP-1640-B00A1 for the green LED, which is diffuse, and HLMP-NG07 for the red and yellow LEDs, which are water clear. Is that up to date? IMO they should all have a diffuse lens.