AllYarnsAreBeautiful/ayab-hardware #58: 3 pin connectors in v1.5 have wrong pinout

t0mpr1c3, 2025-05-09:
Comparing the schematic for v1.5 to that of the EMSL interface (https://github.com/AllYarnsAreBeautiful/ayab-hardware/blob/main/ayab-interface/schematic/ayab_rs.pdf):

v1.5 JP7 is equivalent to EMSL J6B.
EMSL J6B has pinout 1=+5V, 2=GND, 3=EOL_R
v1.5 JP7 has pinout 1=GND, 2=EOL_R, 3=+5V (wrong)

v1.5 JP7A is equivalent to EMSL J6C.
EMSL J6C has pinout 1=EOL_R, 2=GND, 3=+5V
v1.5 JP7A has pinout 1=GND, 2=EOL_R, 3=+5V (wrong)

v1.5 JP2 is equivalent to EMSL J7A/J2B.
EMSL J7A/J2B have pinout 1=+5V, 2=GND, 3=EOL_L
v1.5 JP2 has pinout 1=EOL_L, 2=GND, 3=+5V (wrong)

v1.5 JP2A is equivalent to EMSL J2C.
EMSL J2C has pinout 1=EOL_L, 2=GND, 3=+5V
v1.5 JP2A has pinout 1=EOL_L, 2=GND, 3=+5V (correct)

jonathanperret, 2025-05-10:
Ooh, good catch!
Here's a comparison with the 1.4 schematic and the Interface schematic:

| 1.4 | 1.5 | Interface |
|-|-|-|
|[image]|[image]|[image]|
|[image]|[image]|[image]|

Obviously the JP7/JP7A footprints are wrong in 1.5, although since they are only ever used with an extension cable, the pin order could in theory be fixed in the cable. By the way, as Victoria mentioned on Discord, footprint JP7A shouldn't even exist, it's never used to fit a connector on the board.

JP2 on the 910 is the one that uses a cable, so likewise having it flipped is not a huge problem. But it is a problem for the 930.

It is interesting to note that Brother flipped the 2mm-pitch connectors on the KH965/KH900, except for those going to the solenoids.

Once the pin order for JP7 and JP2 is fixed, some thought could be given to connector orientation. In the 1.4 shield, the 2mm-pitch connectors are in two different orientations, which makes assembly more error-prone: 

[image]

The Interface on the other hand has all the connectors facing the same side, which is nice:

[image]

t0mpr1c3, 2025-05-11:
Yes, this is a nice feature of the EMSL shield. It would be nice to have the chips orientated the same way too.

I'm not sure whether there would space to do it on the through hole shield without going to a 4 layer PCB. It's worth thinking about, but perhaps it might be considered a feature rather than a fix and could be postponed for v1.5.

t0mpr1c3, 2025-05-11:
If I understand Viktoria right, she is saying that neither JP7 nor JP7A ever need to be populated so we only need one set of 3 holes to solder the ribbon cable into. I guess that is true, unless you wanted a shield that worked for both KH965 and KH900 machines, in which case you might want to solder 2 cables, one for each type of connector.

t0mpr1c3, 2025-05-11:
If we could get rid of JP7A, I wonder if there would be room to lay the resistors down flat. I just don't like them sticking up unless they are covered in heat shrink.

jonathanperret, 2025-05-11:
> Yes, this is a nice feature of the EMSL shield. It should be possible -- have a look at [`smd`/arduino_shield_smd/arduino_shield_smd.kicad_pcb](https://github.com/AllYarnsAreBeautiful/ayab-hardware/blob/smd/arduino_shield_smd/arduino_shield_smd.kicad_pcb?rgh-link-date=2025-05-11T00%3A17%3A02.000Z)

Not sure what I was meant to look at in that project but it is indeed very easy to flip the connector, just did a quick and dirty test and there's plenty of room to route the traces the other way:

|before|after|
|-|-|
|[image]|[image]|

jonathanperret, 2025-05-11:
> If I understand Viktoria right, she is saying that neither JP7 nor JP7A ever need to be populated so we only need one set of 3 holes to solder the ribbon cable into. I guess that is true, unless you [...]

Yes, JP7 is only ever used to solder wires. Here are pictures of the three variants with connectors populated, from the soldering instructions document:

|machine|shield|
|-|-|
|KH910/KH950|[image]|
|KH930/KH940|[image]|
|KH900/KH965|[image]|

As you can see JP7/JP7A aren't ever populated with a connector. On the 910/950 they stay unused (those signals are in the 10-pin JP5) and on other machines an extension cable goes there.

If you wanted a shield that can be moved between a 930/940 and a 900/965 (by the way, careful with the nomenclature here, the KH900 is an actual model, one of the last to come out and it uses the same 2mm connectors as the 965) I guess you would populate both sets of connectors and solder two cables into JP7? Even if you put two sets of holes there for that case, there is no reason for one of the footprints to be 2mm-pitched. And maybe it would be smarter to solder just one cable to the board and have it split at the other end. Quite a niche scenario in any case IMHO.

jonathanperret, 2025-05-11:
> If we could get rid of JP7A, I wonder if there would be room to lay the resistors down flat. Either that or put heat shrink in the BOM. I just don't like them sticking up unless they are covered in heat shrink. The resistors are vulnerable to getting knocked over and the unprotected leads can cause shorts.

If we need more space on the board, I guess there's always the buzzer connector that can go — I heard Viktoria stopped packaging buzzers with her kits anyway, probably because user feedback was unanimous that the sounds from the app in 1.0 were so much better than the awful bleating from that electromechanical alarm buzzer.

I would have liked to avoid creep in these updates though… really the only critical thing that motivated an update to the (through-hole) shield was the right-side sensor fix for the 910, and then there's the move from Eagle to KiCad. But that kind of creates a context in which it's tempting to slip in every change we can think of, even though each of those needs to be tested, documented, and so on which can snowball into getting stuck with lots of unproven changes (to wit: this issue) and the project in a state of not being able to produce working hardware.

In my mind we could have just changed that couple traces required for the right-side fix, and we wouldn't even have had to redo all the pictures in the soldering instructions documents. But I also appreciate that stuff is being improved, I just hope we can soon get to a stable state again.

t0mpr1c3, 2025-05-11:
I am sensitive to that. IMO the change to Kicad was always going to mean
redoing the pictures. It can wait but FWIW don't think laying the resistors
flat would require documenting or testing.

jonathanperret, 2025-05-11:
> I am sensitive to that. IMO the change to Kicad was always going to mean
> redoing the pictures. It can wait but FWIW don't think laying the resistors
> flat would require documenting or testing.

I can agree that it would not require _more_ documentation or testing than what we're in for anyway with the changes already done, so I'd be fine with tweaking the resistor footprints if there is space on the board — not that I'm in a position to decide for or against anyway.

jonathanperret, 2025-05-12:
Here's another point I just thought of regarding connector orientation: it would be good to keep the ability to use angled connectors on the edge of the board, I think Victoria uses those for the 965.

Looking at the current shield on `main`, it would be a problem:

[image]

Based on the shield pictures above, it seems the angled versions of the connectors have their "cut corners" on the bottom. With the solenoid connectors as they are now, they would face towards the inside of the shield.

jonathanperret, 2025-05-12:
> That's a fair point. But to be completely honest, I don't trust the pictures in the soldering document. Either the "1" is at the wrong end on the v1.4 silkscreen, or the connectors on the left hand side of the board are back to front.

OK, ignoring that photo, here's a [Molex datasheet](https://www.mouser.com/catalog/645/usd/1640.pdf):

[image]

Regardless of pin 1's position, these drawings clearly show that the angled version of the header has the corner cutouts on the bottom. So, with the way the connectors are set up on the 1.5 shield currently:

[image]

substituting an angled header will not work. And given that these are obsolete parts I think it's not a bad idea to keep options like that open.

In fact now that I think about it, you're probably right that the problem is the `1` on the 1.4 silkscreen. Indeed, comparing the 1.4 schematic to the Interface schematic one can see that the pin numbers are reversed on the shield. There's even a note on the Interface schematic about this!

| 1.4 shield | 1.5 shield | Interface |
|-|-|-|
|[image]|[image]|[image]|

I guess the connector pinout wasn't looked at closely on the shield as it was always easy to orient the connector whichever way worked — and the initial versions used only unpolarized headers. And I imagine it's a coincidence that the correct orientation worked out for angled headers, but as I said above I think that's a useful property to keep.

t0mpr1c3, 2025-05-12:
OK, that makes sense. Is the orientation of the right angle KH930-style connectors also this way round? That is to say, with pin 1 on the right hand side when the wires are pointing down.

t0mpr1c3, 2025-05-12:
Never mind, I just had a look at the Molex 5267 series and I can't find any right angle connectors.

jonathanperret, 2025-05-12:
> Never mind, I just had a look at the Molex 5267 series and I can't find any right angle connectors.

Oh they exist, they just have a different series number. I think it is 5268 in that case. Yep, looking at the datasheet for the plug you can see both 5267 and 5268 are the mating parts: https://www.molex.com/en-us/products/part-detail/50375083.

To answer your question, yes it looks like they are numbered the same:

[image]

[image]

 Pin 1 is on the left side looking into the connector when the cut corners are on the bottom (which coincides with the wires pointing down, in the case of an angled connector).

ETIBerlin, 2025-05-12:
Concerning straight or angled 8 or 10 pin connectors, the angle saves bending the cable from the machine by 90 degrees, but in practices there is no difference if straight or bent. It doesnt really affect how much space it takes up or how it fits into the machine....therefore imo negligable.
When looking at how the shield is fitted into the machine (the connectors facing upwards) then the previous orientation in 1.4 of the connectors is best, as you dont have to twist the cables to fit it to the board. Since the interface fits in the machine like the original board (connectors facing down) it makes sense they have the same orientation as the original hardware which is opposite to shield 1.4.

closed 2025-05-13 by t0mpr1c3

jonathanperret, 2025-05-13:
> When looking at how the shield is fitted into the machine (the connectors facing upwards) then the previous orientation in 1.4 of the connectors is best, as you dont have to twist the cables to fit [...]

Interesting! Some people do put the shield the other way though? See https://www.thingiverse.com/thing:3281342/files for example.