AllYarnsAreBeautiful/ayab-hardware #60: 2mm pitch 5 pin connector has wrong pinout in v1.5

t0mpr1c3, 2025-05-11:
It's back to front.

From EMSL interface schematic:

[image]

From v1.5 schematic:

[image]

jonathanperret, 2025-05-11:
I guess it was already the case in the 1.4 shield, but without a pin 1 mark on that 2mm footprint, one just had to know the connector needed to be flipped, as documented in the soldering instructions:

[image]

Like discussed on #58, having a consistent connector orientation would sure be better.

t0mpr1c3, 2025-05-11:
Yes, the error in the schematic can be resolved by flipping the connector.

closed 2025-05-13 by t0mpr1c3