AllYarnsAreBeautiful/ayab-hardware #67: Options for I2C header

t0mpr1c3, 2025-05-15:
Background:
Currently both the shield and the EMSL interface use 2.54mm pin header with a 5V-SDA-SCL-GND pinout. The ESP prototype uses a 1mm pitch JST SH connector with GND-3.3V-SDA-SCL pinout, that is identical to the Sparkfun QWIIC specification. This pinout is the closest thing that exists to a standard, and has been also adopted by Adafruit Stemma, which uses 2mm pitch JST PH connectors, and Seeed Studio's Grove, which uses proprietary connectors with 2mm pitch. QWIIC uses only 3.3V, whereas both Stemma and Grove allow 5V lines and signals.

Problem:
The 2.54mm pitch header on the shield and interface has the advantage of being generic and robust. The downsides are:
* The connector is somewhat bulky for something that carries low voltage, low current signals. 
* The arbitrary pinout requires documentation, preferably on the PCB itself, not least because...
* ...the header is non-polarised, which means that it is possible to accidentally plug in a matching socket back to front.
These are all good reasons to adopt one of the more standard Qwiic/Stemma/Grove connectors instead. 

The ESP prototype has opted for a Qwiic-compatible header, which is perfectly reasonable given that the ESP32 is powered by 3.3V. However, this makes it somewhat unsuitable for the shield and interface, which employ 5V microprocessors which would require level shifting to output 3.3V signals. In addition, the 1mm pitch hardware used by Qwiic is fiddly and less than robust for an external connection.

Suggested solution:
Ideally we would standardise on a single connector that could be used by all the AYAB hardware. Neither the pin header used by the shield nor the Qwiic connector chosen for the ESP prototype seems to fit the bill. Of the alternatives, both 2mm pitch connectors, Stemma is preferable to Grove because it uses generic connectors that are cheap and universally available.

t0mpr1c3, 2025-05-16:
A slight correction to the above: while it is true that there has always been a through hole version of the shield -- and there is a strong consensus that there should continue to be one -- there have in the past been versions of the shield that have incorporated SMD components, e.g. v1.2 https://github.com/AllYarnsAreBeautiful/ayab-hardware/blob/6542afcab6f2a81585d12e9e20f1e5dc20c1d853/Shieldv1x/10_ayab-arduino/smd/export/arduino_shield.pdf

jonathanperret, 2025-05-16:
Thanks for the recap @t0mpr1c3 .

Regarding the choice of connector for the ESP32 board, if I remember correctly the consensus reached at the last meeting, we agreed to keep the ESP32 for networking only and essentially mirror the Arduino+shield hardware on the board.

With that in mind I think it would make sense to have a 5V I2C connector on the ESP32 board, connected to the 5V MCU used for low-level I/O. And of course we would use the same connector as on the shield — I favor Stemma as well (JST PH 2mm).

That connector would probably be used for an automatic color changer once sufficient energy is gathered to bring that project to fruition.

Note that this would not preclude having an additional Qwiic (3.3V, 1mm) connector directly connected to the ESP32, to connect a display module for example.

t0mpr1c3, 2025-05-26:
By the way the EMSL interface with Metro Mini v2 has a Stemma QT I2C connector built in, as well as the pin header